The-Bug-Bashers/iBOARCode #4: add handling of multiple WebSocket's

Flottegurke, 2025-03-09:
there are now 2 web sockets (command, data) but the logic of the web site (for the connect and disconnect) is only build for one. 

To-Do:

- [ ] rewrite connect and disconnect logic to allow for connection and disconnection of individual web sockets

Flottegurke, 2025-03-09:
note: to test, the api can be hoasted locally from the IDE once maven build scripts got executed. once the api is hoasted, the website will automatically connect to the right ip and port

dagadeta, 2025-03-09:
Should all WebSocket messages be displayed in the console?

Flottegurke, 2025-03-09:
no, only the ones from the command WebSocket, because the data WebSocket will send 7 Lidar data messages and 2 motor state messages per second

Flottegurke, 2025-03-09:
until the lidar and motor rendering scripts are implemented, it would be great if all messages form the /data WebSocket would get logged in the browser console (not the self implemented one)

Flottegurke, 2025-03-09:
Donne in: 8176a4a

closed 2025-03-09 by Flottegurke