The-Bug-Bashers/iBOARCode #7: Make entire controll panel scrollable

Flottegurke, 2025-03-11:
Since there are multiple screen ratios and resolutions, it would be greate, if the entire frontend was scrolable

dagadeta, 2025-03-11:
@Flottegurke 
I assume you mean every section being scrollable in itself, not the entire page?

Flottegurke, 2025-03-12:
I actually meant the entire page, (you now 2 scowl bar at the right which scoos everything) but making every section scrollable by it self is interesting to.

I thought the entire page, because it is eayser to implement with the screen ratios, and it would be quite difficult to make 3very section by itself scolsble, because you would need to insert black space at the bottom iv every section in oder for those people who can't se the bottom, to be able to scroll up enough

Flottegurke, 2025-03-14:
Done with 08de7ba4611e808c95f1031dbd4d36b7b5353c4e

closed 2025-03-14 by Flottegurke