The-Bug-Bashers/iBOARCode #24: chat messages

grafchen, 2025-03-29:
the sent messages in the webui are displayed wrong. @dagadeta knows what i mean

Flottegurke, 2025-03-29:
@grafchen for the sake of me nowing what problem @dagadeta is probably hesitant to fix, would you care to elaborate further?

Flottegurke, 2025-03-29:
also: is this importent for the first release of the bot? because if it is, it should get added to the milestone

dagadeta, 2025-03-29:
It's just about fixing the alignment problem with long sent messages (they take up more space to the left for some reason)

Flottegurke, 2025-03-29:
oh, jeah ok, this is not that important btu still good to get donne

Flottegurke, 2025-03-29:
i had a look into it, and found the issue:
the messages that are received have little bit of an margin -left.
the messages that get send have marginn left: auto.
thie means that if the need to line wrap, there is no margin left unlike the reseived messages
this means the fix is setting an min margin left or geting rid of the margin left for reseived messages