The-Bug-Bashers/iBOARCode #25: add settings mode

Flottegurke, 2025-03-29:
the new mode is useful for settings that apply to the web client only.

it should include:

- [ ] (toggle + value): lock LiDAR zoom.
 - [ ] value: between 1 and 1400, unit: cm
- [ ] (toggle): hide motor data

Flottegurke, 2025-03-29:
@dagadeta to aceve the action of the lock LiDAR zoom toggle have a look in the lidarSection.js line 28: `const maxDistance = Math.max(...data.map(point => parseFloat(point.distance)));`

max distance represents the maximal displayed distance in meters. 
So, if the toggle is activated this should get set to the specifyed value (in cm) / 100

dagadeta, 2025-03-29:
There already is an "Enable Motors" button in the header. What should the hide motor data toggle do differently?

Flottegurke, 2025-03-29:
the hide motor data should stop the data from the motors from appearing on the motors display
the enable motors disables the abbility foall programms to move the motors

dagadeta, 2025-03-29:
Ah, you're right...

dagadeta, 2025-04-01:
The settings mode should be finished although it was not tested with iBOAR yet.

Flottegurke, 2025-04-01:
tested, and corrected formular: 844244b0e918c66791afd6ae44ea0d0eac1ee008

closed 2025-04-01 by Flottegurke