The-Bug-Bashers/iBOARCode #29: keydown event listeners not getting removed

Flottegurke, 2025-04-05:
the key down event listeners for the turning buttons in the remote control made seem to not get removed causing 1 keypress to trigger multiple messages after repeatedly opening the remote control mode.

dagadeta, 2025-04-06:
Isn't this frontend and with that could be assigned to me?

Flottegurke, 2025-04-06:
yeah is is,I don't now what happened here

dagadeta, 2025-04-07:
For the "w", "a", "s", and "d" keys, the event listeners are added only once because the `handleKeyDown` and `handleKeyUp` functions check the keyState object to ensure that the key press is handled only once. The "q" and "e" keys are added multiple times. This can be prevented by checking if the event listeners have already been added before adding them again.

dagadeta, 2025-04-07:
Fixed with 0f1fb24c354cdb5699639fedf29eed449935de80

closed 2025-04-07 by dagadeta